openrobotics/ros-central-registry #9: Convert liburcu from rules_foreign_cc to native bazel

asymingt, 2025-09-09:


tfoote, 2025-11-12:
That's great its updated upstream one less thing to maintain. 

While he's away this is mostly going to be in maintenance. You should be able to fork it and change the submodule pointer for testing yourself. 

The intent is to evolve the system to store the patches in the registry instead of in forks. But we haven't gotten there yet so things are not as maintainable as we'd like. Since this is still a testing/prototype we aren't planning to have specific maintenance coverage through the end of the year after which asymingt should be back.

asymingt, 2025-12-05:
Thanks for flagging this, @wep21! I'd like to retain the BCR naming to avoid confusion. What do you think about this? https://github.com/intrinsic-opensource/ros-central-registry/pull/87